mischa85/Ozzy #2: No issue, but great work

andy83-projects, 2024-05-23:
I don't image that ploytec made multiple dfferent soundcards for A&H

What's the usb device id for the DB4?  Curious if this same driver would work for my Xone:4D (looking fwd to macos version)

mischa85, 2024-05-28:
The USB ID of the Xone:DB4 is 0x0A4A:0x0FFD.

I actually have a Xone:4D too, but it's power supply is broken. I can imagine this could work on the 4D with minimal changes.

ppkdsvbs, 2024-08-19:
Hi @mischa85, great work!

Do you still need Wireshark data from XONE 4D? I just bought 4D and can install Windows 10 to generate data for you. Let me know.

mischa85, 2025-02-03:
Hi there,

Since you posted this issue I updated the driver to support several other devices. Including the 4D.

andy83-projects, 2025-04-23:
I gave it a go on a spare Mac I have.  Seems to work just as well as the Ploytec revival.

great work.

mischa85, 2026-10-11:
Closing: this was about the old DriverKit app, which has been replaced by the kext + HAL driver (`macos/install.command`). Thanks for testing! If the current version gives you trouble, please open a new issue.

closed 2026-10-11 by mischa85