mischa85/Ozzy #4: This is my feedback and thank you for your work

gamingoxygen, 2024-07-12:
Hardware : 
Xone DB4 plus two Xone K2
MacBook Pro 15 Mid 2014 Intel + Nvidia
This setup work perfect whit macOS 10 .15 Catalina and official drivers.

My mac support only macOS 11 Big Sur (Official)
1 . I compile the app  on  macOS 11 Big Sur xcode 13.2 successfully
2 . Surprise the app is work only whit macOS 12 Monterey and UP
3 . I saved the app and i install macOS 12 Monterey whit  OpenCore Legacy Patcher
4 . When i prepare macOS 12 Monterey whit  OpenCore Legacy Patcher  i Disabled SIP and AMFI in security settings
5 . Now the xonedb4 app is work and i cant install the driver
6. I boot in recovery mode and  use ( csrutil disable ) whit succes the second comand (sudo nvram boot-args="amfi_get_out_of_my_way=0x1") fail  whit sudo in recovery mode and i try whit no sudo 
7. Reboot the sistem and again i cant install the driver  Error: Make sure SIP is disabled (csrutil disable in recovery) nand amfi_get_out_of_my_way=0x1 is added to the bootflags
8. No successful installation to test the driver

It is possible to have a ready compile app or .dmg file or more version of them whit support from macOS 11 Big Sur and up 12,13....? 
it is possible to have a separate compile app or .dmg for intel and arm ?
and the app don't have a logo maybe the xone logo is a good idea.
Im happy to test the driver if i can install first and give you feedback

this is my feedback thank you for your work and sorry for my English

mischa85, 2024-07-12:
Thank you for testing! As far as I know AudioDriverKit (which all of this is based on) is, according to the Apple docs, only available on macOS 12.1+ (Monterey and up). But I see that the RME DriverKit drivers also work on macOS 11 (Big Sur), so this might actually work.

In normal macOS startup, check if AMFI is disabled in the boot-args in terminal with: `nvram boot-args`
And check the SIP status with: `csrutil status`
And check if developer mode is on with: `systemextensionsctl developer`

These three things should make sure you'll be able to load the self-signed driver.

mischa85, 2026-10-11:
Closing: this was about the old DriverKit app, which has been replaced by the kext + HAL driver (`macos/install.command`). Thanks for testing! If the current version gives you trouble, please open a new issue.

closed 2026-10-11 by mischa85