mischa85/Ozzy #5: DB2 compatibility

WeDigJames, 2024-08-19:
Your contribution is really appreciated, especially as Allen & Heath said it couldn't be done.

I have a DB2. I guess the sound card is the same. Do you think your work would be compatible or require much tweaking to make it work for the DB2?

Thank you!

mischa85, 2024-08-19:
I also think they are pretty similar. I will need a Wireshark USB capture of your DB2 running on the latest audio firmware (this is the Ploytec updater thingy distributed along the driver). Capture USB traffic using Wireshark, plug in the mixer, play some audio for a few seconds and stop Wireshark.

WeDigJames, 2024-09-09:
I managed to do the Wireshark capture for my DB2.
[Here is the file](https://www.dropbox.com/scl/fi/jfslowwd09y8clcjrg9us/xone_DB2.pcapng?rlkey=bi9d3a36thhf671wads57gg20&st=15gweqyl&dl=0).
I hope that works - my first time using Wireshark. A lot of data there for a few seconds.

mischa85, 2024-09-10:
Some observations:

- The capture looks absolutely identical to what I saw on my DB4 running on old audio firmware. It's using the old BULK transfers for PCM in/out and MIDI out. Can you double check you have it running on the latest 1.2.0 audio firmware? You find this audio firmware with the audio fwupdater in the "XONE:DB2 Device Software", please don't confuse it with the mixer firmware "XONE:DB2 Firmware".
- The capture looks fine, but it's missing important bits being transferred the moment the mixer is connected. Please start the capture before you plug in the mixer. You can stop capture shortly after.
- Please compress the file before uploading.

All in all, looks like it'll be easy to add support for the DB2.

mischa85, 2024-09-10:
Thanks!

This gives me the USB ID I'll need, which is 0a4a:ffd2
```
<key>idVendor</key>
<integer>2634</integer>
<key>idProduct</key>
<integer>65490</integer>
```

It also tells me that the mixer runs on audio firmware `1.0.8`. Do you happen to have an older macOS or Windows machine on hand to update the firmware to `1.2.0`?

WeDigJames, 2024-09-10:
Thanks for sticking with me!

Dusted off on old laptop running Sierra and got 1.2.0 installed on the mixer.
So, [here's the new Wireshark](https://www.dropbox.com/scl/fi/jzu2xg6sgo2ebacpp6uyb/xone_DB2.pcapng.zip?rlkey=081kkzpfp32ki88iynoq8dakm&st=fc2np10m&dl=0), hopefully with everything you need to take a look.

Really appreciate your help with this, thanks.

mischa85, 2024-09-11:
No worries, thanks for gathering the data.

Your new capture looks identical to what I see on my DB4. So much even that I think the driver should work if you only change the USB ID:

https://github.com/mischa85/snd-xonedb4/blob/cb43ec01a5ec87355d9ff1f1714748c45ba025c2/mac-coreaudio/XoneDB4Driver/Info.plist#L51

to:

`<integer>65490</integer>`

mischa85, 2025-01-04:
Hi there, I just added DB2 support to the Linux driver.

WeDigJames, 2025-01-09:
Good to know, thanks (though I am a Mac user).

I do currently have a paid for Apple developer account. I used it for a project that didn't end up going anywhere. It expires in April, but I thought I could offer its use to make the signed binary.

I don't imagine that I'll renew it - I don't know if the binary will need to be renewed annually - and I'd need some support to use the account to do what's necessary, but I can offer it if you can make use of.

mischa85, 2025-01-09:
Also added support in the macOS driver today: https://github.com/mischa85/snd-xonedb4/commit/1d8ae43f2923d200393c8ee68bb2b6bb853bedbb

If you could compile the driver and post a signed binary that would be amazing. Testing would be a whole lot easier if people don't need to navigate compiling and circumventing the system integrity and code signing hurdles Apple put up.

WeDigJames, 2025-01-09:
I'll see what I can do, though I fear there may be many questions generated.
Would you be up for joining me on a screen sharing session sometime to go through the process? May be we stand a better chance of getting it done together.

mischa85, 2025-01-09:
For sure. But before we can get anything signed you'll need to request the entitlements from Apple. Could you try the get the DriverKit entitlements via the link?

The information they'll need:

- The transport mechanism for your hardware. DriverKit supports USB, PCI, HID, networking, and serial devices.
**USB**

- Your company’s hardware vendor ID for the type of devices.
**0x0A4A**

- Brief descriptions of the devices.
**Audio & MIDI interface**

- A brief description of the app you'll use to install the driver.
**GUI DriverKit installation and monitoring**

mischa85, 2026-10-10:
DB2 support should now work on macOS and Linux as of ce19792 (correct packet layout, 48 kHz). If anyone with a DB2 can confirm, I'll close this.