mischa85/Ozzy #9: Windows Driver - Xone 4D

RuRate13, 2024-11-07:
Is it also usable for the 4D and Windows? And if so, and the file is included.... How do I install the driver on Windows? Sorry if the question is obvious...

jobermayr, 2025-01-17:
Currently there isn't support for Xone:4D at all.

@mischa85: Here is some basic information. Please write a private email (you obfuscate yours really good) if you need some wireshark and/or Int964 encoder/decoder from Mac/Win driver):

lsusb:
```
Bus 001 Device 003: ID 0a4a:ff4d Ploytec GmbH Allen&Heath Xone:4D
Negotiated speed: High Speed (480Mbps)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0a4a Ploytec GmbH
  idProduct          0xff4d Allen&Heath Xone:4D
  bcdDevice            1.00
  iManufacturer           1 Ploytec GmbH
  iProduct                2 Allen&Heath Xone:4D
  iSerial                 3 no serial
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0049
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            5
          Transfer Type            Isochronous
          Synch Type               Asynchronous
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
[... 226 lines cut ...]
00000168	23.09807968	AsyncPort::stopRead	
00000169	23.09808540	PGDevice::stopStream called from PGDevice::deviceUnplugged	
00000170	23.09808731	KernelThreadBank::setRtsFunc 0000000000000000	
00000171	23.09808731	PGDevice::setStreamingStarted to 0 from PGDevice::stopStream. Count:6	
00000172	23.19834328	PGDevice::setStreamingStarted to 0 from PGDevice::stopAudioRequests. Count:7	
00000173	23.19834900	PGDevice::stopAudioRequests countUsed:0	
00000174	23.19835281	PGDevice::stopStream called from PGKernelDevice::abortPipes	
00000175	23.19835472	KernelThreadBank::setRtsFunc 0000000000000000	
00000176	23.19835663	PGDevice::setStreamingStarted to 0 from PGDevice::stopStream. Count:8	
00000177	23.19836044	AsyncPort::stopRead	
00000178	23.19836235	AsyncPort::clearEndpoints	
00000179	23.19836235	AsyncPort::stopRead	
00000180	23.19836235	InterruptAudio::onUnconfigure	
00000181	23.19836426	InterruptAudio::onUnconfigure	
00000182	23.19837952	PtSubDevice::term FFFFBE8465C3CB40	
00000183	23.19850540	PtSubDevice::term FFFFBE845FDD61B0	
00000184	23.19910622	exit KernelThreadBank::Thread::loop()	
00000185	23.20409775	AsyncPort::term	
00000186	23.20410156	AsyncPort::stopRead	
00000187	23.20411873	InterruptAudio::term	
00000188	23.20467949	InterruptAudio::term	
00000189	23.20604515	UNLOAD XONE:4D USB ASIO driver (USB) 2.9.91.0 built:Feb  7 2018 - 21:16:38	
00000190	23.20605087	Core::term	
00000191	23.20605469	Core::threadsTerm 0000000000000000	
```

mischa85, 2025-01-17:
@jobermayr Thanks for the information! I've seen enough Wireshark captures to safely assume the 4D talks the same way as the other "USB2 Audio Launchpad"-based devices. Support added in:

https://github.com/mischa85/snd-xonedb4/commit/3fde285c7f3d20026dfd5a64633e794ea393867c
https://github.com/mischa85/snd-xonedb4/commit/876ad99c0dee309f58e04526c304a1a768abe1e0

mischa85, 2025-01-17:
Hmmm, I expect a couple of URB fails when the device resets. A device reset is necessary as we can only set the hardware sample rate that way.

Is PCM working?

jobermayr, 2025-01-17:
No. Device shows up but don't work. I really don't understand that communication kernel<>usbcode<>device but I assume it "shuts down" because transferred data is maybe wrong. Original win driver has sth. like:
```
struct DMAChannel {
  int left;
  int right;
};

struct DMAPackage {
  DMAChannel channels[5];
};

struct DMAPackages {
  DMAPackage package[20];
};

struct AudioBulkIRPBlock {
  int unknown[512];
};

struct Encoding {
  DMAPackages* packages;
  AudioBulkIRPBlock* block;
  int* unknown;
};

// Simplified by AI
int32_t dmaEncode10_AH_Int964_C(Encoding* src) {
  if (src == nullptr || src->packages == nullptr) {
    return 800; // Error handling
  }

  int8_t* dst = reinterpret_cast<int8_t*>(src->block); // Use block for output

  for (int i = 0; i < 20; ++i) { // Iterate through packages
    for (int j = 0; j < 5; ++j) { // Iterate through channels
      int* data = reinterpret_cast<int*>(&src->packages->package[i].channels[j]); 
      for (int k = 0x1E; k > 6; --k) {
        *dst++ =
            ((data[3] >> (k - 2) & 1) << 3) |
            ((data[0] >> (k + 1) & 1) << 0) |
            ((data[4] >> (k - 3) & 1) << 4) |
            ((data[2] >> (k    ) & 1) << 1) |
            ((data[1] >> (k - 1) & 1) << 2);
      }
    }
  }

  return 800; // Success
}
```

mischa85, 2025-01-17:
Can you send a Wireshark USB capture of both the working Windows driver and the non-working Linux driver? Plugout/in, some audio, some MIDI... Then I'll see if I can figure out what's going wrong.

marcel@hackerman.art

mischa85, 2025-01-24:
I'm working on this one. I suspect it has something to do with the way the driver tries to reset the device using `usb_reset_device`. I tried using `usb_lock_device_for_reset` before, but this results in a deadlock on my system.