mischa85/Ozzy #10: Issue When Opening the Application

zedmithat, 2024-11-26:
I encounter this error when I open the application. I would appreciate your assistance.
[image]

2024-11-26 22:55:40.286449+0300 XoneDB4App[1926:65741] system extension didFailWithError: The operation couldn’t be completed. (OSSystemExtensionErrorDomain error 8.)

mischa85, 2025-02-03:
Hi there,

This means the app is not signed. Can you try again with a recent version of the code?

mischa85, 2026-10-11:
Closing: this is about signing the old DriverKit version, which no longer exists. The current macOS driver (kext + HAL, `macos/install.command`) doesn't need DriverKit entitlements. If the current version doesn't work, please open a new issue with the `debuglogs.command` output.

closed 2026-10-11 by mischa85