mischa85/Ozzy #11: Merry Christmas - almost there?

surfacescan, 2024-12-26:
Hey there - I am running the built app and I see the following.  It all looks good until I click the Open User client button.

I ran the log and got the following:
`Timestamp                       Thread     Type        Activity             PID    TTL
2024-12-25 19:05:45.471536-0500 0x3e86     Default     0xec43               1292   0    XoneDB4App: system extension actionForReplacingExtension: <private> <private>
2024-12-25 19:05:45.522811-0500 0x3e86     Default     0xec43               1292   0    XoneDB4App: system extension didFinishWithResult: 0`

[image]

mischa85, 2025-01-04:
Hi there. What happens if you press the "Open User Client" button?

To be honest, the development of the macOS driver has been incredibly frustrating due to issues I explained on the README. Anwyays, should be possible to make it work.

Can you try to retrieve some more logging using:
`log stream --predicate 'sender == "sc.hackerman.xonedb4driver.dext"' --predicate 'sender == "XoneDB4App"' --predicate 'sender == "kernel"'`

surfacescan, 2025-01-04:
Hey there - I tried the Open User Client button and nothing happened.  What should I see?

 - here is the output from the log command:
 - 
[image]

Any ideas?

I tried the "Uninstall Dext" at one point and then aborted because it said it wanted to change system level extensions... that may be some of the errors in the log

Anything else I could try?  I think I successfully followed the other steps.

mischa85, 2025-01-04:
Couple of observations:

- For some reason macOS wants the DriverKit binary in arm64e format. This is a result of `amfi_get_out_of_my_way=0x1` in `boot-args`. You need to change the `Architectures` for the project to `arm64e`. `ARCHS = "$(ARCHS_STANDARD)"` becomes `ARCHS = arm64e`.
- Some logs are showing <private> and hiding the real errors. Installing [this](https://georgegarside.com/blog/macos/sierra-console-private/) profile fixed that for me.

surfacescan, 2025-01-04:
It's ALIVE!  It sounds SOOOOOOOOO good! 

[image]
[image]

Thanks - I did exactly what you said and changed the architecture in the XCode Project file as so:

[image]

![Alt Text] (https://i.giphy.com/media/v1.Y2lkPTc5MGI3NjExdm53dGR5ZXR4aWtsbm01d3RzemFndG5wY3Byam4yM3hnaGQydm1ycCZlcD12MV9pbnRlcm5hbF9naWZfYnlfaWQmY3Q9Zw/l2JhrFaYBW5ovrTnW/giphy-downsized-large.gif)

Since I am guessing that many of the people who have tried to get this working may have got lost in the software engineering parts of this, I would be happy to create a bit more of a "I like to mix music and don't know anything about compiling software" installation guide if you want!?

Thank you!

surfacescan, 2025-01-04:
Wait - so I don't need to run the XoneDB4 app to use the mixer? I just realised this!

mischa85, 2025-01-04:
Glad to hear!

> Since I am guessing that many of the people who have tried to get this working may have got lost in the software engineering parts of this, I would be happy to create a bit more of a "I like to mix music and don't know anything about compiling software" installation guide if you want!?

Indeed. The problem you encountered is something I can fix in code tho. Will do. But also many people get lost in code signing issues. Of course: things would be better if I could just make a signed driver available which would avoid the steps required to compile the driver and get macOS to actually accept it. But the only option I found is paying for the Apple Developer Program which I kindly refuse.

mischa85, 2025-01-04:
Build targets updated in code:

https://github.com/mischa85/snd-xonedb4/commit/be47352874cec83bce5660819333454732d40b0b

OneTenEleven, 2025-01-08:
I also seem to be lost in Code Signing issues! 
If you have a "I like to mix music" install guide @surfacescan that would be amazing! 

I currently have these errors:

XoneDB4Driver isn't code signed but requires entitlements. It is not possible to add entitlements to a binary without signing it.

system extension didFailWithError: Missing entitlement com.apple.developer.system-extension.install

I am not familiar with Xcode at all, but would love to get the DB4 working on my M4 MBP! 
Also considering a Ploytec Revival, but came across this driver so wanting to try and get this running.

Cheers.

mischa85, 2025-01-08:
I will try my best to write up things in a bit more detail in the README. But as you might have guessed I prefer coding and getting things to work to making guides. Maybe you missed changing the CODE_SIGN_IDENTITY in codesign.sh?

Change it for the value you find using `security find-identity`.

mischa85, 2025-01-08:
https://github.com/mischa85/snd-xonedb4/commit/9b8c1edfb870212c003c7b153129ae8456b5826b

surfacescan, 2025-01-09:
hey @mischa85 - I have gotten the driver working but it is a bit hit and miss. I can do a more thorough breakdown of the scenarios that I lose connection to the mixer if that would be helpful.\

 But before that, I want to rule out a possible culprit. I am also encountering some challenges now on my Digitakt which I connect through USB to something called Overbridge (a plugin that lives in Ableton and allows 8 virtual tracks).  

Someone from Elektron thought that perhaps the `nvram` settings and the `csrutil ` I did to get the XoneDB4 driver running might be contributing to problems with the Digitakt.  Is it safe to change these settings "back" now that I have the app signed?

If yes - what might those commands be?

I wonder if this might help make the mixer more stable for my general use on my powerbook.

After a bit more testing, I would be happy to help write a more "I just wanna mix" step by step guide to getting things up and running.

mischa85, 2025-01-09:
Hey @surfacescan, first of all, thank you for testing. It's motivating te keep developing when people are actually testing the driver.

Interesting case. I know the Digitakt is also using Ploytec software/hardware and I know they've been working on a DriverKit driver. But sadly Elektron never released any source code so I could never have a look. They have also been battling macOS security mechanisms: https://support.elektron.se/support/solutions/articles/43000566545-blocked-overbridge-driver-macos-

On top of that, we're also battling signing issues here. I will experiment as well, but can you what prevents the driver to load and function? You can reverse the changes like this:

- Reboot the system and keep ```COMMAND ⌘ + R``` pressed while booting, this will bring you in recovery.
- Open a terminal.
- ```csrutil enable``` to enable System Integrity Protection.
- ```sudo nvram boot-args=""```
- Reboot to macOS.
- Open a terminal.
- ```systemextensionsctl developer off``` to disable developer mode.

OneTenEleven, 2025-01-09:
Thanks for updating the README @mischa85. 

I made some progress last night after following the new guide and managed to briefly connect the DB4. 
Tested with Traktor & Ableton and just got hiss. 

Unfortunately the driver seems unstable and after a while I get an error:

"XoneDB4Driver has experienced an error during deactivation.
Please check the logs to find the error."

When opening in Xcode I see this in the Build:
/Library/Developer/Xcode/DerivedData/XoneDB4Driver-bjyblkzasgzrcuayklqbutmdjill/Build/Intermediates.noindex/XoneDB4Driver.build/Debug/XoneDB4App.build/Objects-normal/arm64e/XoneDB4AppViewModel.o' was not built with class_ro_t pointer signing enabled, but previous .o file was not

And 

Initialization of immutable value 'outdiff' was never used; consider replacing with assignment to '_' or removing it
Initialization of immutable value 'indiff' was never used; consider replacing with assignment to '_' or removing it

I copied the full 40 digit code from terminal into codesign.sh

XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX "Apple Development: ajkitson@gmail.com (XXXXXXXXXX)" 

Let me know if I am missing anything! 

@surfacescan Happy to help out with a guide too if I get this running...

mischa85, 2025-01-09:
> Tested with Traktor & Ableton and just got hiss.

My first thought is that your DB4 firmware is still using the BULK transfer mode (pre 1.4.1 firmware). I actually just re-added support for this transfer mode to the Linux driver. I'll add it to the macOS driver too. Some 2012-2013ish Macbooks have a bug in the USB controller requiring this update for smooth audio streaming, but some devices never received it.

But in your case: please update the audio firmware.

OneTenEleven, 2025-01-09:
I just checked the DB4 firmware and I am running v2.2.2

I tested on both M1 Max & M4 Max MBP running macOS Sequoia 15.2 and found the same issue.

mischa85, 2025-01-09:
You are confusing the mixer firmware with the audio interface firmware. The audio firmware updater is part of the "XONE:DB4 Device software": https://www.allen-heath.com/hardware/legacy-products/xonedb4/resources/

OneTenEleven, 2025-01-09:
Ah ok, I had not installed this on the M series as thought your driver was replacing this. I will test and get back to you....

mischa85, 2025-01-09:
You don't need to install the driver though. There's a separate executable in there that allows you to update the audio firmware.

mischa85, 2025-01-09:
Yep, that's where the problem is. The FWUpdater has never been compiled for arm64 (Apple silicon) and needs to be reverse engineered as well.

If you have an older Mac or any Windows machine around you can run it on there. I think the Windows driver even has the update to 1.4.1 in there.

OneTenEleven, 2025-01-09:
Ah ok, so no chance on the M chip Macs right now? 
What are the chances of reverse engineering this?!

mischa85, 2025-01-09:
I'm actually working on adding support for the old BULK transfer mode to the macOS driver. Just like what I did to the Linux driver: https://github.com/mischa85/snd-xonedb4/commit/df71ad7578525f491dd3c0d67b5836d742abbba5. That should make DB4/DB2's running on old firmware work.

Again, if you have any Windows computer around just for the update, that'll make the driver work properly on your Apple silicon Mac.

OneTenEleven, 2025-01-09:
Ok, fingers crossed then! 

I have no Windows machine, I could dig out an old Mac though!

mischa85, 2025-01-09:
Okay I re-added bulk streaming to the macOS driver. Needs testing, might have some math wrong.

https://github.com/mischa85/snd-xonedb4/commit/6be3dbee9f77516c42471db9dce907b111d5726d

OneTenEleven, 2025-01-09:
Nice work! I have clean Audio coming out of the DB4 from Ableton! 
I also have Traktor Scratch control working :)
I will investigate Midi mappings next...

OneTenEleven, 2025-01-09:
Ableton doesn't seem to recognise the DB4 for Midi mapping. 
I will did out my DB4 Traktor mapping and test Midi control later....

mischa85, 2025-01-09:
Nice to hear audio is working!

MIDI is at the moment only implemented in the Linux driver. I'll see if I can get some support in the macOS driver asap.

OneTenEleven, 2025-01-09:
Good stuff. I will keep an eye on development and test further once you have MIDI implemented.

mischa85, 2026-10-11:
Closing: this was about the old DriverKit app, which has been replaced by the kext + HAL driver (`macos/install.command`). Thanks for testing! If the current version gives you trouble, please open a new issue.

closed 2026-10-11 by mischa85