mischa85/Ozzy #12: MacOS Monterey Xone DX testing

sidgallup, 2025-01-18:
I did some more testing installing/uninstalling, plugging/replugging, heres the log, one observation tho, trying to uninstall the driver always gives an error, its ays something about checking the error log to see what the error was.
This testing was done on 2 different MacOS installs, same machine but different hard drives and OS versions, one has Monterey 12.7.4 the other Monterey 12.7.6, i can install Sonoma on one of the hard drives  to test if you think this would be useful or will make a difference in any way.

sid@SidGallups-MacBook-Pro ~ %
log stream --predicate 'sender == "sc.hackerman.xonedb4driver.dext"' --predicate 'sender == "XoneDB4App"' --predicate 'sender == "kernel"'
Filtering the log data using "sender == "sc.hackerman.xonedb4driver.dext" OR sender == "XoneDB4App" OR sender == "kernel""
Timestamp                       Thread     Type        Activity             PID    TTL  
2025-01-18 08:40:46.377963-0600 0x30cf     Default     0x0                  0      0    kernel: plugin-container[841] triggered unnest of range 0x7ff84b600000->0x7ff84b800000 of DYLD shared region in VM map 0xfb83f901c347818b. While not abnormal for debuggers, this increases system memory footprint until the target exits.
2025-01-18 08:41:03.004224-0600 0x2d4a     Default     0x8c2a               817    0    XoneDB4App: system extension actionForReplacingExtension: <private> <private>
2025-01-18 08:41:03.201648-0600 0x2d4a     Default     0x8c2a               817    0    XoneDB4App: system extension didFinishWithResult: 0
2025-01-18 08:41:11.630931-0600 0x3167     Default     0x0                  0      0    kernel: DK: XoneDB4Driver-0x1000009df waiting for server sc.hackerman.xonedb4driver-1000009df
2025-01-18 08:41:11.639488-0600 0x316e     Default     0x0                  0      0    kernel: sc.hackerman.xonedb4driver[844] Corpse allowed 1 of 5
2025-01-18 08:41:11.710149-0600 0x68       Default     0x0                  0      0    kernel: Corpse released, count at 0
2025-01-18 08:41:11.710164-0600 0x3167     Default     0x0                  0      0    kernel: DK: XoneDB4Driver-0x1000009df user server timeout
2025-01-18 08:41:11.888192-0600 0x116d     Default     0x0                  0      0    kernel: uipc_accept: peer disconnected unp_gencnt 1312
2025-01-18 08:41:28.807524-0600 0x31c7     Default     0x0                  0      0    kernel: DK: XoneDB4Driver-0x1000009f1 waiting for server sc.hackerman.xonedb4driver-1000009f1
2025-01-18 08:41:28.815176-0600 0x31cd     Default     0x0                  0      0    kernel: sc.hackerman.xonedb4driver[846] Corpse allowed 1 of 5
2025-01-18 08:41:28.842842-0600 0x68       Default     0x0                  0      0    kernel: Corpse released, count at 0
2025-01-18 08:41:28.842875-0600 0x31c7     Default     0x0                  0      0    kernel: DK: XoneDB4Driver-0x1000009f1 user server timeout
2025-01-18 08:41:35.467888-0600 0x374      Default     0x0                  0      0    kernel: IOConsoleUsers: time(0) 0->0, lin 1, llk 0,
2025-01-18 08:41:35.467909-0600 0x374      Default     0x0                  0      0    kernel: IOConsoleUsers: gIOScreenLockState 1, hs 0, bs 0, now 0, sm 0x0
2025-01-18 08:41:36.352088-0600 0x323d     Default     0x0                  0      0    kernel: plugin-container[848] triggered unnest of range 0x7ff84b600000->0x7ff84b800000 of DYLD shared region in VM map 0xfb83f901c347deeb. While not abnormal for debuggers, this increases system memory footprint until the target exits.
2025-01-18 08:41:37.878858-0600 0x374      Default     0x0                  0      0    kernel: IOConsoleUsers: time(0) 0->0, lin 1, llk 0,
2025-01-18 08:41:37.878887-0600 0x374      Default     0x0                  0      0    kernel: IOConsoleUsers: gIOScreenLockState 1, hs 0, bs 0, now 0, sm 0x0
2025-01-18 08:41:38.399487-0600 0x2d4a     Default     0x8f1b               817    0    XoneDB4App: system extension didFinishWithResult: 0
2025-01-18 08:41:40.755747-0600 0x2d4a     Default     0x9030               817    0    XoneDB4App: system extension requestNeedsUserApproval
2025-01-18 08:41:46.290834-0600 0x2d4a     Default     0x869d               817    0    XoneDB4App: system extension didFinishWithResult: 0
2025-01-18 08:41:53.278472-0600 0x32cb     Default     0x0                  0      0    kernel: DK: XoneDB4Driver-0x100000a15 waiting for server sc.hackerman.xonedb4driver-100000a15
2025-01-18 08:41:53.326409-0600 0x32cf     Default     0x0                  0      0    kernel: sc.hackerman.xonedb4driver[851] Corpse allowed 1 of 5
2025-01-18 08:41:53.396855-0600 0x68       Default     0x0                  0      0    kernel: Corpse released, count at 0
2025-01-18 08:41:53.396893-0600 0x32cb     Default     0x0                  0      0    kernel: DK: XoneDB4Driver-0x100000a15 user server timeout
2025-01-18 08:42:02.165021-0600 0x3308     Default     0x0                  0      0    kernel: DK: XoneDB4Driver-0x100000a28 waiting for server sc.hackerman.xonedb4driver-100000a28
2025-01-18 08:42:02.172554-0600 0x330c     Default     0x0                  0      0    kernel: sc.hackerman.xonedb4driver[852] Corpse allowed 1 of 5
2025-01-18 08:42:02.201554-0600 0x68       Default     0x0                  0      0    kernel: Corpse released, count at 0
2025-01-18 08:42:02.201556-0600 0x3308     Default     0x0                  0      0    kernel: DK: XoneDB4Driver-0x100000a28 user server timeout
2025-01-18 08:42:09.011489-0600 0x75       Default     0x0                  0      0    kernel: PMRD: power event 14 args 0x0 0x0
2025-01-18 08:42:09.011510-0600 0x75       Default     0x0                  0      0    kernel: PMRD: wrangler RootDomainPower (ps 4, 0->1)
2025-01-18 08:42:09.011513-0600 0x75       Default     0x0                  0      0    kernel: PMRD: evaluatePolicy( 0, 0x0 )
2025-01-18 08:42:09.025471-0600 0x75       Default     0x0                  0      0    kernel: PMRD: power event 14 args 0x0 0x0
2025-01-18 08:42:09.025489-0600 0x75       Default     0x0                  0      0    kernel: PMRD: wrangler RootDomainPower (ps 4, 0->1)
2025-01-18 08:42:09.025492-0600 0x75       Default     0x0                  0      0    kernel: PMRD: evaluatePolicy( 0, 0x0 )
2025-01-18 08:42:09.054897-0600 0x3332     Default     0x0                  0      0    kernel: memorystatus: set assertion priority(0) target AvatarPickerMemojiPicker:797
2025-01-18 08:42:09.085550-0600 0x75       Default     0x0                  0      0    kernel: PMRD: power event 14 args 0x0 0x0
2025-01-18 08:42:09.085576-0600 0x75       Default     0x0                  0      0    kernel: PMRD: wrangler RootDomainPower (ps 4, 0->1)
2025-01-18 08:42:09.085579-0600 0x75       Default     0x0                  0      0    kernel: PMRD: evaluatePolicy( 0, 0x0 )

mischa85, 2025-01-22:
Is the driver matching with the device? As in, does anything change/appear in logs when plugging in the device?

Maybe it's interesting to also log the `kernelmanagerd`, to see what happens:

`log stream --predicate 'sender == "sc.hackerman.xonedb4driver.dext"' --predicate 'sender == "XoneDB4App"' --predicate 'sender == "kernel"' --predicate 'sender == "kernelmanagerd"'`

sidgallup, 2025-01-26:
Yes i see some text appears on terminal when i plug and replug the DX, im doing all this with your app open, and it never detects it since i can see the red text an the veruy bottom reading NOT CONNECTED.

This is the new log, im plugging and replugging the unit a few times and seeing if it gets detected by your app but still no dice.

Last login: Sun Jan 26 12:04:51 on console
sid@Sids-MacBook-Pro ~ % log stream --predicate 'sender == "sc.hackerman.xonedb4driver.dext"' --predicate 'sender == "XoneDB4App"' --predicate 'sender == "kernel"' --predicate 'sender == "kernelmanagerd"'
Filtering the log data using "sender == "sc.hackerman.xonedb4driver.dext" OR sender == "XoneDB4App" OR sender == "kernel" OR sender == "kernelmanagerd""
Timestamp                       Thread     Type        Activity             PID    TTL  
2025-01-26 12:05:56.905142-0600 0x174a     Default     0x4e64               628    0    XoneDB4App: system extension actionForReplacingExtension: <private> <private>
2025-01-26 12:05:56.987806-0600 0x168d     Default     0x4e64               108    0    kernelmanagerd: sysextd delegate validate extension: sc.hackerman.xonedb4driver (1.0/1)
2025-01-26 12:05:56.988131-0600 0x168d     Default     0x4e64               108    0    kernelmanagerd: [com.apple.kernelmanagerd.logging:DextValidation] Validating dext  0x0p+0t sc.hackerman.xonedb4driver
2025-01-26 12:05:56.988599-0600 0x168d     Default     0x4e64               108    0    kernelmanagerd: [com.apple.kernelmanagerd.logging:DextValidation] Realize dext sc.hackerman.xonedb4driver
2025-01-26 12:05:57.005275-0600 0x168d     Default     0x4e64               108    0    kernelmanagerd: validating extension at /Library/SystemExtensions/.staging/3D4B9A20-B3C9-4335-97C4-E19461FE44B0/sc.hackerman.xonedb4driver.dext
2025-01-26 12:05:57.013172-0600 0x168d     Default     0x4e64               108    0    kernelmanagerd: sysextd delegate will-replace extension: sc.hackerman.xonedb4driver (1.0/1) with extension: sc.hackerman.xonedb4driver (1.0/1)
2025-01-26 12:05:57.014035-0600 0x168d     Default     0x4e64               108    0    kernelmanagerd: symlink /Library/DriverExtensions/sc.hackerman.xonedb4driver.dext -> /Library/SystemExtensions/3D4B9A20-B3C9-4335-97C4-E19461FE44B0/sc.hackerman.xonedb4driver.dext
2025-01-26 12:05:57.017425-0600 0x168d     Default     0x4e64               108    0    kernelmanagerd: sysextd delegate will-terminate extension: sc.hackerman.xonedb4driver (1.0/1)
2025-01-26 12:05:57.021425-0600 0x168d     Default     0x4e64               108    0    kernelmanagerd: sysextd delegate terminate extension: sc.hackerman.xonedb4driver (1.0/1)
2025-01-26 12:05:57.028931-0600 0x168d     Default     0x4e64               108    0    kernelmanagerd: sysextd delegate will-start extension: sc.hackerman.xonedb4driver (1.0/1)
2025-01-26 12:05:57.029537-0600 0x168d     Default     0x4e64               108    0    kernelmanagerd: sysextd delegate start extension: sc.hackerman.xonedb4driver (1.0/1)
2025-01-26 12:05:57.029933-0600 0x168d     Default     0x4e64               108    0    kernelmanagerd: symlink /Library/DriverExtensions/sc.hackerman.xonedb4driver.dext -> /Library/SystemExtensions/3D4B9A20-B3C9-4335-97C4-E19461FE44B0/sc.hackerman.xonedb4driver.dext
2025-01-26 12:05:57.030392-0600 0x168d     Default     0x4e64               108    0    kernelmanagerd: [com.apple.kernelmanagerd.logging:DextValidation] Realize dext sc.hackerman.xonedb4driver
2025-01-26 12:05:57.047354-0600 0x168d     Default     0x4e64               108    0    kernelmanagerd: validating extension at /Library/SystemExtensions/3D4B9A20-B3C9-4335-97C4-E19461FE44B0/sc.hackerman.xonedb4driver.dext
2025-01-26 12:05:57.049667-0600 0x168d     Default     0x4e64               108    0    kernelmanagerd: [com.apple.kernelmanagerd.logging:KernelClient] Loading codeless extension: Dext sc.hackerman.xonedb4driver v1 in executable dext bundle sc.hackerman.xonedb4driver at /Library/SystemExtensions/3D4B9A20-B3C9-4335-97C4-E19461FE44B0/sc.hackerman.xonedb4driver.dext
2025-01-26 12:05:57.062951-0600 0x174a     Default     0x4e64               628    0    XoneDB4App: system extension didFinishWithResult: 0
2025-01-26 12:06:03.382867-0600 0x512      Default     0x0                  0      0    kernel: uipc_accept: peer disconnected unp_gencnt 697
2025-01-26 12:06:05.350923-0600 0x17b7     Default     0x0                  108    0    kernelmanagerd: Received kext load notification: sc.hackerman.xonedb4driver
2025-01-26 12:06:05.350736-0600 0x17b6     Default     0x0                  0      0    kernel: DK: XoneDB4Driver-0x10000074b waiting for server sc.hackerman.xonedb4driver-10000074b
2025-01-26 12:06:05.364748-0600 0x17ba     Default     0x0                  0      0    kernel: sc.hackerman.xonedb4driver[634] Corpse allowed 1 of 5
2025-01-26 12:06:06.050015-0600 0x68       Default     0x0                  0      0    kernel: Corpse released, count at 0
2025-01-26 12:06:06.050025-0600 0x17b6     Default     0x0                  0      0    kernel: DK: XoneDB4Driver-0x10000074b user server timeout
2025-01-26 12:06:17.506052-0600 0x181c     Default     0x0                  0      0    kernel: necp_process_defunct_list: necp_update_client abort nexus error (2) for pid 485
2025-01-26 12:06:17.506460-0600 0x75       Default     0x0                  0      0    kernel: PMRD: power event 14 args 0x0 0x0
[... cut ...]
2025-01-26 12:06:58.973077-0600 0x1918     Default     0x0                  0      0    kernel: PMRD: setAggressiveness(0) kPMMinutesToSleep = 10
2025-01-26 12:06:58.973971-0600 0x75       Default     0x0                  0      0    kernel: PMRD: aggressiveness changed: system 0->10, display 10
2025-01-26 12:06:58.973983-0600 0x75       Default     0x0                  0      0    kernel: PMRD: idle time -> 0 ms (ena 1)
2025-01-26 12:06:58.973986-0600 0x75       Default     0x0                  0      0    kernel: PMRD: idle sleep timer enabled

mischa85, 2025-01-26:
Hmmm. I wanna rule out anything to do with the support of the other devices. I made a new branch which only supports the Xone:DX. Can you try again using this branch?

`git clone http://github.com/mischa85/snd-xonedb4 -b xonedxtest`

sidgallup, 2025-01-31:
Still no luck, nothing has changed... heres a log i captured with the new version build you posted above. Am i supposed to install your dext everytime i run your app? because everytime i open it it says "driver deactivated" and i have to click on install again.
 Im not sure if im doing this correctly, so heres what im doing exactly to test your app.

move app to Apps folder, open it, click on install dext, security tells me to go to system prefs to allow the app to install, i do it, click on install dext again and this time your app tells me it is installed, then i plug my DX unit and no change in the app, only some text on terminal, i unplugg and replugg 3 times, still no dice.

Last login: Fri Jan 31 17:46:22 on console
sid@Sids-MacBook-Pro ~ % log stream --predicate 'sender == "sc.hackerman.xonedb4driver.dext"' --predicate 'sender == "XoneDB4App"' --predicate 'sender == "kernel"' --predicate 'sender == "kernelmanagerd"'
Filtering the log data using "sender == "sc.hackerman.xonedb4driver.dext" OR sender == "XoneDB4App" OR sender == "kernel" OR sender == "kernelmanagerd""
Timestamp                       Thread     Type        Activity             PID    TTL  
2025-01-31 17:50:20.067301-0600 0xfc7      Default     0x0                  108    0    kernelmanagerd: Received kext load notification: com.apple.driver.AppleUSBAudio
2025-01-31 17:50:20.067637-0600 0xfc7      Default     0x0                  108    0    kernelmanagerd: Received kext load notification: sc.hackerman.xonedb4driver
2025-01-31 17:50:20.067492-0600 0x1ec1     Default     0x0                  0      0    kernel: DK: XoneDB4Driver-0x100000784 waiting for server sc.hackerman.xonedb4driver-100000784
2025-01-31 17:50:20.078046-0600 0x1ec3     Default     0x0                  0      0    kernel: sc.hackerman.xonedb4driver[709] Corpse allowed 1 of 5
2025-01-31 17:50:20.833515-0600 0x68       Default     0x0                  0      0    kernel: Corpse released, count at 0
2025-01-31 17:50:20.833521-0600 0x1ec1     Default     0x0                  0      0    kernel: DK: XoneDB4Driver-0x100000784 user server timeout
2025-01-31 17:50:34.444800-0600 0x1f0e     Default     0x0                  0      0    kernel: DK: XoneDB4Driver-0x10000079b waiting for server sc.hackerman.xonedb4driver-10000079b
2025-01-31 17:50:34.444796-0600 0x1f0f     Default     0x0                  108    0    kernelmanagerd: Received kext load notification: sc.hackerman.xonedb4driver
2025-01-31 17:50:34.452688-0600 0x1f12     Default     0x0                  0      0    kernel: sc.hackerman.xonedb4driver[711] Corpse allowed 1 of 5
2025-01-31 17:50:34.483037-0600 0x68       Default     0x0                  0      0    kernel: Corpse released, count at 0
2025-01-31 17:50:34.483059-0600 0x1f0e     Default     0x0                  0      0    kernel: DK: XoneDB4Driver-0x10000079b user server timeout
2025-01-31 17:50:42.671298-0600 0x75       Default     0x0                  0      0    kernel: PMRD: power event 14 args 0x0 0x0
2025-01-31 17:50:42.671317-0600 0x75       Default     0x0                  0      0    kernel: PMRD: wrangler RootDomainPower (ps 4, 0->1)
2025-01-31 17:50:42.671320-0600 0x75       Default     0x0                  0      0    kernel: PMRD: evaluatePolicy( 0, 0x0 )
2025-01-31 17:50:52.782482-0600 0x1f82     Default     0x0                  0      0    kernel: DK: XoneDB4Driver-0x1000007b5 waiting for server sc.hackerman.xonedb4driver-1000007b5
2025-01-31 17:50:52.782447-0600 0x1f84     Default     0x0                  108    0    kernelmanagerd: Received kext load notification: sc.hackerman.xonedb4driver
2025-01-31 17:50:52.790369-0600 0x1f86     Default     0x0                  0      0    kernel: sc.hackerman.xonedb4driver[712] Corpse allowed 1 of 5
2025-01-31 17:50:52.820244-0600 0x68       Default     0x0                  0      0    kernel: Corpse released, count at 0
2025-01-31 17:50:52.820249-0600 0x1f82     Default     0x0                  0      0    kernel: DK: XoneDB4Driver-0x1000007b5 user server timeout
2025-01-31 17:51:06.371423-0600 0x75       Default     0x0                  0      0    kernel: PMRD: power event 14 args 0x0 0x0
2025-01-31 17:51:06.371443-0600 0x75       Default     0x0                  0      0    kernel: PMRD: wrangler RootDomainPower (ps 4, 0->1)
2025-01-31 17:51:06.371447-0600 0x75       Default     0x0                  0      0    kernel: PMRD: evaluatePolicy( 0, 0x0 )

mischa85, 2025-02-01:
I had a Monterey machine on hand and was able to reproduce the issue.

I fixed it in the main branch: https://github.com/mischa85/snd-xonedb4/commit/406ac4abbd11dc473de166bc4d8a5c5c3dec8381

For some reason this was only a problem on older macOS versions.

Thank you for your patience.

sidgallup, 2025-02-03:
great to be helping debug this project. So, should i rebuild the main app again? following the steps on the tutorial?

mischa85, 2025-02-03:
Yes, that should work.

sidgallup, 2025-02-03:
duuuuuudeee yes, its alive now. Been playing some tunes for about an hour sounds great no hiccups, its only crashed on me twice, complete freeze and had to shut down the laptop manually pressing power button.

Great work, now lets see if adding MIDI support is even possible. Will test this version of your app this week, to report how stable it is. Quick question tho, about the buffer size, is there a size you recommend or can i just chose freely? on Traktor Pro 4 have it set at 256,  app crashed on me at 256 and then 512, im not sure if i have to change the settings on your app so theres no conflict.

mischa85, 2026-10-11:
Closing: this was about the old DriverKit app, which has been replaced by the kext + HAL driver (`macos/install.command`). If the DX gives trouble with the current version, please open a new issue with the `debuglogs.command` output.

closed 2026-10-11 by mischa85