mischa85/Ozzy #13: Beginner having problems compiling

Freeminder23, 2025-01-24:
I'm kinda new to this "linux thing" but would love to keep using my beloved DB:04 Mixer. I have resently installed Linux Mint 22.1 Kernel: 6.8.0-51-generic arch: x86_64 and got several issues while compiling your driver. Could fix the missing "flex not found" and "bison not found" errors on my own (with the help of the internet), but now I'm stuck with following: 

make[3]: *** No rule to make target 'arch/x86/entry/syscalls/syscall_32.tbl', 
needed by 'arch/x86/include/generated/uapi/asm/unistd_32.h'. Exit.
make[2]: *** [arch/x86/Makefile:249: archheaders] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: exiting from directory “/usr/src/linux-headers-6.8.0-51-generic”
make: *** [Makefile:5: linux] Error 2 

Could your please be so kind to help me fix this? Big thank you in advance
for providing this work to others to help keep this exelent hardware allave and connected.

Freeminder23, 2025-01-25:
It seemed to work till the end, but when I do the "modprobe snd-usb-xonedb4" I get a
 "modprobe: ERROR: could not insert 'snd_usb_xonedb4': Operation not permitted"
and if I sudo  "modprobe snd-usb-xonedb4" a
"modprobe: ERROR: could not insert 'snd_usb_xonedb4': Exec format error"

And the cinnamon sound-control applet has now disappeared, which is a bit unfortunate, I would needed that back.

mischa85, 2025-01-25:
This could happen when you forget to reboot after a kernel update (you are still on the old kernel and the module is expecting the new kernel).

mischa85, 2026-10-11:
Closing, since the answer above should solve it and the Linux build has been restructured since. Please open a new issue if current `main` doesn't build.

closed 2026-10-11 by mischa85