mischa85/Ozzy #14: No consigo armar el controlador !!!!

J-Karbonell, 2025-02-03:
Hola, quería exponeros mi problema a ver si me podeis ayudar... ya que me viene esto un poco grande...
E seguido todos los pasos, pero llega el momento de clonar y me da ERROR... esto es lo que me dice:
 
xcrun: error: invalid active developer path (/Library/Developer/CommandLineTools), missing xcrun at: /Library/Developer/CommandLineTools/usr/bin/xcrun

E intentado buscar la forma de arreglarlo por internet, pero llevo un par de semanas y no se que hacer... 

Un saludo y gracias.

sidgallup, 2025-02-03:
tienes instalado Xcode? que versiond e macOS y que version de Xcode estas usando?

mischa85, 2025-02-03:
Hola hermanos.

Please use English on GitHub.

closed 2025-02-03 by mischa85