mischa85/Ozzy #15: Add support for Dynacord CMS 600-3

mischa85, 2025-02-05:
The original variant of this mixer uses a Ploytec audio interface.

idVendor: `0x0562`
idProduct: `0x03eb`

| Endpoint | Transfertype | Packetsize | Purpose |
| - | - | - | - |
| 0x00 (OUT) | x | x | control |
| 0x81 (IN) | ISO | 64 | unknown |
| 0x02 (OUT) | ISO | 156 | unknown |
| 0x03 (OUT) | BULK | 512 | MIDI? |
| 0x04 (OUT) | BULK | 512 | unknown |
| 0x86 (IN) | BULK | 512 | PCM |

More information at:
https://github.com/dominics/dynacord-cms-pcap

martinspinler, 2026-02-04:
Hi!

I have Alesis Multimix 8 USB 2.0, which has a similar USB endpoint configuration.
I did some research and experiments, and would like to share my knowledge.

According to the datasheet, the Dynacord has 4 In / 4 Out Audio USB channels.
Alesis has 10 In / 2 Out channels (no MIDI).

The recorded PCAP uses 96kHz sample rate.

Audio In
========

The Bulk In endpoint 0x86 seems to encode Audio In data in the same way as decoded by `ploytec_convert_to_s24_3le`:
URB is sent each 3 ms, 18432 B per URB = 288 frames per URB (64B per frame).

My Alesis has bit different parameters: Sends URB with length 131072 B = 2048 frames.

Audio Out
=========

The Audio Out goes through ISO Endpoint 0x02.
URB is sent each 3 ms. For 96 kHz, it should be the same: 288 frames per URB.
URB has average data length 3456 B, which gives 3456 / 288 = 12 B per frame.
It seems to be stored in raw PCM, and can be decoded like this:

```python
import itertools
urb_data = [...]
nr_in_channels = 4 if 'DYNACORD' else 2 if 'ALESIS' else -1
for frame in itertools.batched(urb_data, n=3 * nr_in_channels):
    for channel, sample in enumerate(itertools.batched(frame, n=3)):
        pcm_value = sample[0] + (sample[1] << 8) + (sample[2] << 16)
        if pc_value >= 2**23:
           pc_value -= 2**24
```

The ISO flow creates 24 packets in one URB with average 12 frames per packet.
(This makes sense, USB 2.0 packet rate is 8000 p/s, 8000 * 12 = 96kHz.)

Average frame rate
------------------

The average frame rate is important as the frame rate varies for different sample rates.

My Alesis has a bit different parameters:
Each 5 ms, an URB with 40 packets is transmitted.
For 44100 Hz sample rate, the average frame rate is 5.5125 (44100 * 0.005 / 40), which is
approximated like this: [5, 6, 5, 6, 6, 5, 6, 5, 6...] (in frames per packet). Similar pattern is shown in my PCAP.

Sync In
=======

There is an ISO endpoint 0x81 for some synchronization.
It is pulled every 5 ms (both Dynacord and Alesis) by one URB with 5 packets. Each packet holds 3 bytes.
(The pcap/Wireshark says that the time from request is 45ms, which probably means there is 8-10 submitted URBs in queue from driver.)

I think that one packet contains the number of processed frames (sent to Audio In) per 1ms.
One packet holds the newest value and two historical values.
However, I've experienced that some packets can be missing (urb->iso_frame_desc[i].actual_length != 3) or frames skipped.

From my point of view is like this:

```
URB | Packet 0 | Packet 1 | Packet 2 | Packet 3 | Packet 4 |
0   | F0 xx xx | F1 F0 xx | F2 F1 F0 | F3 F2 F1 | F4 F3 F2 |
1   | F5 F4 F3 | II II II | F7 F6sF4 | F8 F7 F6 | F9 F8 F7 |
...

- Fn means number of frames in period n
- xx means begin of stream (contains 0, assuming not valid values)
- II means missing packet
- s  means skipped frame(s)
```

I think the Sync In is in original driver primarily used for low-latency audio input (URB for Audio In on my Alesis is 2048 frames, ~21ms), but should be used for syncing output too.
 
Current state of my experiments
===============================

I managed to implement the ISO URB handling for Audio Out and for Sync In in Linux driver.
For the sample rate 48000Hz (gives exactly 6 frames per packet), I was able to generate likely clean sine wave (in driver only; no ALSA yet).
But without proper synchronization, the USB rate desyncs with the Mix rate after few seconds, and the buffer underruns/overruns occurs on the Mix and clicking/distortion appears in the sound. I've experimented with different sync mechanisms, but none of them works yet.

For the sample rate 44100 Hz, the sine wave out is very distorted from the very beginning. I don't know, what is so wrong.

mischa85, 2026-11-01:
The Ploytec revision of the Dynacord CMS 600-3 (`0562:03EB`) is now in Ozzy as an experimental device (f8b241a on `main`), on Linux and macOS.

Credit to @dominics: everything came from your work in [dominics/Ozzy](https://github.com/dominics/Ozzy) and [dominics/dynacord-cms-pcap](https://github.com/dominics/dynacord-cms-pcap). The descriptors, feedback packets and handshake in your captures showed it's the same layout as the Akai EIE Pro, which Ozzy already drives. So it's a device-table entry on the existing isochronous path, not new code:

- 4 in / 4 out at 44.1, 48 and 96 kHz
- MIDI on the first port (the DIN sockets). The second port ("SystemCtrl", FX control) isn't offered yet.
- Rate changes select both interfaces again before the handshake, which your driver found the CMS needs (Ozzy now does this for every Ploytec device).

The newer class-compliant revision needs no driver. In case it shares the same USB ID, Ozzy now leaves any listed device whose interface 0 is USB Audio class to the system's class driver, on Linux and macOS. Windows can't tell the two apart, so don't install Ozzy on Windows for a class-compliant unit.

It hasn't run on a CMS 600-3 yet. If you have one, please install from `main`, play and record at each rate, try MIDI, and post the log (Linux: `sudo dmesg | grep -i ozzy`, macOS: `debuglogs.command`), whether it works or not. The steps are in [Testing an Experimental Device](https://github.com/mischa85/Ozzy#-testing-an-experimental-device). On macOS, Ozzy uses a kernel extension, unlike dominics' userspace driver, so it needs the reduced security setup from the README.

mischa85, 2026-11-01:
@martinspinler Sorry this went unanswered for so long, and thanks for the detailed write-up. It was exactly right, and it's what let me add your MultiMix 8 without having one.

Your analysis of the sync endpoint 0x81 matches what Ozzy does: each 3-byte packet is the frame count per millisecond over the last 3 ms, newest first, and Ozzy paces the isochronous output from it. That pacing is what your driver was missing: without it the output drifts against the mixer's clock, which is why it clicked after a few seconds. At 44.1 kHz the packets then come out as the 5/6 pattern you saw.

The Alesis MultiMix USB 2.0 (`13B2:0030`) is now in Ozzy as an experimental device (4ee0de5 on `main`), on Linux and macOS, on the same path as the Akai EIE Pro and the CMS 600-3:

- 10 in / 2 out, as on your MultiMix 8 (Alesis's own driver claims 4 / 4)
- 44.1 and 48 kHz, the rates Alesis's driver offers (the spec says up to 96 kHz; not enabled until someone tests it)
- no MIDI

Could you try it? Please install from `main` (steps in [Testing an Experimental Device](https://github.com/mischa85/Ozzy#-testing-an-experimental-device)), play and record at both rates, and post `sudo dmesg | grep -i ozzy` and the output of `lsusb -v -d 13b2:0030`, whether it works or not. The descriptors would confirm the endpoints, which I took from your description.

dominics, 2026-11-02:
Hell yeah @mischa85, nice work! I'll rebase my fork on your upstream and test your version of the kext on my CMS 600-3 (have been running my forked version for a week or two and it has been solid!)

Thanks for your initial work that got this all going :+1:

mischa85, 2026-11-14:
Thanks for testing, @dominics, and for pinning the MIDI bug down to the exact lines. You were right: OzzyMIDI's own parser sent `F0` on its own and then dropped the data bytes, and it never reused a running status.

Fixed in 942111e: macOS MIDI input now goes through the same parser the Windows driver uses (`common/midi_ump.c`, with tests for running status and SysEx), and apps get complete messages. Your two sequences now come back as `90 3C 40` / `90 3E 40` / `90 40 40` and as the whole SysEx. That affects every device's MIDI in on macOS, not just the CMS.

Could you rerun your loopback test on macOS with current `main`? `install.command` again, then the same SysEx and running-status sequences.

Also good to see the CMS 600-3 working at all three rates on both systems. Once macOS MIDI checks out, I'll move it from experimental to supported.