mischa85/Ozzy #16: Applications not recognising my db4

nwere, 2025-03-07:
hii thank for this but I am having issues 
Mac - Intel Mac on Sequoia
issue -  Application not recognising my db4 
anyone has more detail instructions on how to do this 
it might be user error 
sorry not a coder

nwere, 2025-03-07:
it says zone db4driver has experienced and error during deactivation.
driver extension is not running

nwere, 2025-03-07:
[image] screenshot of error

mischa85, 2025-03-07:
Can you check the following?

Is System Integrity Protection disabled?
`csrutil status`

Is `amfi_get_out_of_my_way=0x1` in the `boot-args`?
`nvram boot-args`

Did you sign the app with a valid identity?
`security find-identity -v`

nwere, 2025-03-07:
csrutil status - is disabled 
security find-identity -v -      0 valid identities found

mischa85, 2025-03-07:
The `codesign` script will fail if you don't have a valid identity. You can get one via Xcode.

nwere, 2025-07-16:
btw it now says  2 valid identities found

mischa85, 2025-07-17:
Then the codesign script should be able to do it's thing.

closed 2025-07-24 by mischa85