mischa85/Ozzy #23: Assigning channels in Traktor settings/UI update?

alec22ross, 2025-06-12:
Are there any updates regarding being able to assign channels in Traktor/the UI?

mischa85, 2025-06-12:
Hi there,

I think you are talking about MIDI in/out. Whenever I find some free time I try to do some experimenting on the test branch:
https://github.com/mischa85/snd-xonedb4/tree/test

It doesn't work yet as I'm trying to figure out how to get a MIDI message from kernel space to user space. When that works I can implement some mechanism to talk to CoreMIDI. Apple really is nuts.

alec22ross, 2025-06-14:
I guess I'm not sure if my install is working for my Xone DB4. I can't see in Traktor settings the channel assignments. I should be able to right just so i can use it for the sound card? I'm not really planning to assign midi controls from the mixer.

mischa85, 2025-06-14:
Yes, audio should be working.

alec22ross, 2025-06-15:
I can see it as an audio device in Traktor, but I cant see any channels.

mischa85, 2025-06-15:
Is it listed in macOS System Settings?

alec22ross, 2025-06-15:
Yes i believe so. Apologies. When I'm home later today or tomorrow I'll attach some screen shot snips.

alec22ross, 2025-07-11:
[image]
[image]

alec22ross, 2025-07-11:
I keep getting the error that I dont have a developer ID. But I am logged into Xcode.

alec22ross, 2025-07-13:
I ran the git pull and make mac. It doesn't show in apple audio devices but i can see it in device manager.

alec22ross, 2025-07-13:
Can i pay you to do a work window with me remotely so you can configure my set up? I'm dying to use my DB4 with my Kontrol D2s in Traktor 4. Pretty please lolol

alec22ross, 2025-07-13:
> It seems like there is a driver active, as it lists something in Traktor. You're using an old version of the driver here, please update using git pull and make mac.
>
> Are you having trouble signing the driver? Follow this thread:
> #25<https://github.com/mischa85/snd-xonedb4/issues/25>

Yea mischa85, I'm having issues signing the driver. I tried Thead: 25 but no luck. Still shows a personal account in Xcode and no valid developer account in terminal.

If you could show me how to get it all set up via a remote work window that would be super helpful and i can compensate you what you ask.

Regards,

Alec Ross

Datacenter Technician, Blue Ridge Campus C (BL20/BL22/BL31/IAD21/IAD31/WAS05)

Mobile: +1 (703) 254-9838

***@***.******@***.***>

Sun-Wed 7am-7pm | EST

[cid:d8990a47-fc7c-45f7-a3af-3cf5e77c715d][Microsoft Datacenter Operations Services WCS RMA Specialist][A blue circle with white text and a logo  Description automatically generated][A logo with text and a world map  Description automatically generated]

Classified as Microsoft Confidential

mischa85, 2025-07-13:
The main problem is the codesigning failing. That's the reason it complains about it loudly in terminal. macOS refuses to run a non-signed driver, even if you disabled both SIP and AMFI. It just won't work at all.

Did you at the very least try installing the certificate? It seems to be a common problem:
https://www.apple.com/certificateauthority/AppleWWDRCAG3.cer

But: I have no clue why the DB4 does show up in the Traktor window above tho, so something doesn't seem to add up in your story.

mischa85, 2025-07-13:
It shows up under System Information because your system can still detect devices present on the USB bus. Even without a driver.

You really need to fix the codesigning issue. Your system just won't load an unsigned driver. That driver is what will make audio and MIDI work.

You can confirm you have this problem by running:
`security find-identity -v`

Unfortunately, I have not found one single answer to solve this. It is just Apple being stupid. You can Google this problem and find hundreds of people having the same problem.

mischa85, 2026-10-11:
Closing: this was about signing the old DriverKit version, which has been replaced by the kext + HAL driver (`macos/install.command`). Please open a new issue if channel assignment in Traktor doesn't work with the current version.

closed 2026-10-11 by mischa85