mischa85/Ozzy #25: Can´t sign the driver

bjuhasz81, 2025-07-03:
Hello, first of all, thank you very much for making such an effort!

I have a MacBook Air with macOS 15 and can't get any further with compiling the driver. I installed Xcode and also created an account at developer.appple.com. I have agreed to the developer agreement. However, I only have the role of a user in Xcode and not that of a developer. I think this is the reason for the following error message? Sorry for the stupid question, but how do I get the developer role for free so that I can sing the driver?

[image]

Best regards,

bjuhasz81, 2025-07-03:
Thanks for the quick reply. I had looked at Manage Certificates. There is nothing else I can do. My Apple account in Xcode still has the status: Personal Team & User.   

the command: security find-identity -v
delivers: 0 valid identities found

mischa85, 2025-07-03:
DId you try the '+' button in the Manage Certificates dialog?

mischa85, 2025-07-03:
Yep it's probably the Apple certificate chain thing again.

Try the options here and see if anything fixes it:
https://developer.apple.com/forums/thread/123658

bjuhasz81, 2025-07-03:
Thanks, this  ertificate https://www.apple.com/certificateauthority/AppleWWDRCAG3.cer fixed the Problem.

The Driver extension is installed now, but don't recognize my DB2. Another dumb question. Do I need the old A&H Driver to get this work?

[image]

mischa85, 2025-07-03:
No, if that even installs on your mac it will cause conflicts. This is a completely new driver.

It should really be detecting your DB2 though.

bjuhasz81, 2025-07-03:
Finally it the Driver detected my DB2 but Ableton&Serato do not recognize the DB2.

[image]

bjuhasz81, 2025-07-04:
> Hmmm, it doesn't work in any app? Not in browser or music player?

Hi,können wir auf Deutsch schreiben? Nein, es funktioniert weder mit Spotify oder im Browser. Wenn ich was Abspiele, bleibt auch der Status von Playing und Recording auf False. Was mir aufgefallen ist: wenn ich den Mixer abstecke, steht da immer noch connected. Nur die In- und Out-Sample Time erhöht sich dann. Vielen Dank nochmal für Deine Mühe,Grüße Barnabas

mischa85, 2025-07-04:
Please use English on Github as the information should be universally available and understandable.

mischa85, 2025-07-04:
No worries.

Well, something weird is going on. Let's fetch the logging of the driver.

- Unplug the mixer
- Fire up the following in terminal: `log stream --predicate 'sender == "sc.hackerman.ploytecdriver.dext"'`
- Replug the mixer

bjuhasz81, 2025-07-04:
Not the whole thing. Only the last row - wich says "<private>" - is repeated again and again.

mischa85, 2025-07-04:
Those lines will contain clues, but Apple - in their infinite wisdom - tried to make parts of logging private by default.

Try installing the profile from here to see what's being logged:
https://www.cmdsec.com/unified-logs-enable-private-data

mischa85, 2025-07-05:
You can try out the latest build. It could have some more clues about what is going on in the main UI.

- `git pull` to retrieve the latest code
- `make mac` to recompile

bjuhasz81, 2025-07-05:
HI!I have installed the new version of the driver. Playback via DB2 still does not work. BUT: At least I can hear a strong white noise on the outputs of the DB2 after I have connected it and changed the “URB-Count” or “Frames Output/Input” in the driver. The noise also changes the sound when I play something or stop playback.

[image]

[image]

mischa85, 2025-07-05:
Hmmm, both USB PCM Framerate Out and In should be around 96000. This seems seriously off, as if something on the USB chain is messing with the traffic. Are there any USB hubs involved?

mischa85, 2025-07-05:
Any settings that at least do get the USB Framerate to 96000?

bjuhasz81, 2025-07-05:
No. This setting is missing in the driver.

mischa85, 2025-07-05:
Is there any combination of URB count and packet size that gets it to 96000?

bjuhasz81, 2025-07-05:
That's how the Driver looks like, when I just started it up and connected the DB2. Direct connection. No USB-Hubs involved. 

[image]

mischa85, 2025-07-05:
Can you try any of the settings to get the USB going? As in, when you see a value at Framerate In/Out

bjuhasz81, 2025-07-05:
When I start to play around with the URB Count / Frames, then white noise starts. The Framerate IN/OUT are mostly on 48000 or something nearby.

[image]

mischa85, 2025-07-05:
This almost seems like the DB2 fails to set it's samplerate to 96000 Hz. I have no idea why the current code wouldn't succeed in setting that, but I'll do some comparison using the DB2 USB capture I have.

In the meantime, your DB2 runs at 48000 Hz while CoreAudio is set to 96000 Hz. Something you could try is setting CoreAudio to 48000 Hz by changing 96000 to 48000 here:

https://github.com/mischa85/snd-xonedb4/blob/main/mac-coreaudio/PloytecDriver/PloytecDevice.cpp#L72
https://github.com/mischa85/snd-xonedb4/blob/main/mac-coreaudio/PloytecDriver/PloytecDevice.cpp#L77
https://github.com/mischa85/snd-xonedb4/blob/main/mac-coreaudio/PloytecDriver/PloytecDevice.cpp#L87
https://github.com/mischa85/snd-xonedb4/blob/main/mac-coreaudio/PloytecDriver/PloytecDevice.cpp#L110

mischa85, 2026-10-11:
Closing: this is about signing the old DriverKit version, which no longer exists. The current macOS driver (kext + HAL, `macos/install.command`) doesn't need DriverKit entitlements. If the current version doesn't work, please open a new issue with the `debuglogs.command` output.

The DB2 sample rate problem discussed here was fixed in ce19792 (the DB2 runs at 48 kHz only).

closed 2026-10-11 by mischa85