mischa85/Ozzy #27: Support for Allen & Heath Xone:1D

313RGB, 2025-07-07:
I’d like to request support for the Allen & Heath Xone:1D MIDI controller. This is the MIDI only version of the Allen & Heath Xone:2D which already is supported with the Ploytec Driver. As requested via Reddit here is the USB Capture with Terminal:

Xone MIDI Device:

Product ID: 0x001e
Vendor ID: 0x0a4a
Version: 1.04
Speed: Up to 1.5 Mb/s
Manufacturer: Allen & Heath
Location ID: 0x14210000 / 3
Current Available (mA): 500
Current Required (mA): 200
Extra Operating Current (mA): 0

Let me know if there any other info or testing you need me to help with?

mischa85, 2025-07-07:
Hi there!

Low-hanging fruit first: please sent me the output of `lsusb -v`. This includes the endpoint descriptors.

Additionally, I'll need a USB capture of a working system (Windows/older macOS) to see how the device communicates. Use Wireshark, move some knows and trigger some leds so we'll have MIDI in/out.

313RGB, 2025-07-19:
Hi @mischa85 – Thanks, I finally had some time to look at this today. I had to do this via Linux as my Mac/s are all too new OS for Wireshark. 

🛠️ Device Details

Device Name: Allen & Heath Xone:1D
Vendor ID: 0a4a
Product ID: 001e
Bus: 005
Device Address: 011

📦 Technical Data Provided

To assist with implementation, I’ve captured the following:

- lsusb -v -d 0a4a:001e output: [Attached as xone1d-lsusb.txt]
- Full USB traffic capture (MIDI) via Wireshark: [Attached as xone1d-midi.pcapng]
- These are zipped as it doesn't want to upload the .pcapng file?

[Xone 1D Capture July 25.zip](https://github.com/user-attachments/files/21327189/Xone.1D.Capture.July.25.zip)

During the capture, I interacted with the device by:

- Moving all faders and rotary encoders
- Pressing all buttons
- Turning the jog wheel

Feel free to let me know if you need anything else.  Thank you again for your work — I look forward to hearing your thoughts!

mischa85, 2025-07-19:
Thank you! This is already very helpful.

Would you be so kind and also give a USB capture when replugging the device? Also, it would also be helpful if you'd test this on a system that actually functions with the device (the OS sends *and* received the MIDI messages). Is this the case for Linux at the moment?

313RGB, 2025-08-04:
Hi @mischa85 – apologies for the late reply, I did this on the weekend... I’ve now completed the requested testing:

- USB Capture (replugging the device) was done using Wireshark on Linux after loading usbmon. I've included this capture for your analysis, covering the replug event and device enumeration.

- MIDI Monitoring: I booted into a Live Ubuntu USB and used alsa-utils (aseqdump) to check for MIDI input. The device appears in the ALSA sequencer list as Xone:1D MIDI 1, but no MIDI messages are detected when using the knobs or buttons — suggesting that Linux currently does not send/receive MIDI with this device.

For completeness, I also tested on macOS 10.15.7 using MIDI Monitor, but the Xone:1D is not recognised as a MIDI device there either.

✅ I’ve attached:

[Xone 1D Follow Capture August 25.zip](https://github.com/user-attachments/files/21575292/Xone.1D.Follow.Capture.August.25.zip)

- xone1d-usb-replug-capture.pcapng (from Wireshark)
- xone1d-midi-test.log (from aseqdump)

313RGB, 2025-08-05:
Hi @mischa85 – quick follow-up: I’ve re-captured the MIDI testing on Linux, this time by copying the terminal output from aseqdump directly, as the original .log file ended up being empty due to no flushed output.

The new MIDI dump confirms that the Xone:1D is sending MIDI data under Linux — I moved multiple controls (knobs, faders, buttons, jog wheel), and the data is now properly recorded.

✅ Attached:

[Xone-midi-test_UPDATED Aug 25.txt](https://github.com/user-attachments/files/21597085/Xone-midi-test_UPDATED.Aug.25.txt)

- Xone-midi-test_UPDATED Aug 25.txt (copied from aseqdump terminal output)

Let me know if there’s anything else you'd like tested or formatted differently.

mischa85, 2026-10-11:
Sorry for the long silence, and thanks for the captures and dumps. They showed what was going on: the 1D is a standard USB MIDI device, but both of its endpoints use a 4-byte packet size, which the USB spec doesn't allow at its speed. Linux accepts that, macOS doesn't.

- **macOS:** current `main` (9a40a64) adds experimental 1D support. Install with `macos/install.command`; the 1D should then appear as "Xone:1D" in Audio MIDI Setup. Please report whether controls arrive and whether LEDs respond. If it doesn't show up, please attach the output of `sudo dmesg | grep -i ozzy` (or `log show --last 5m --predicate 'sender == "OzzyKext"'`).
- **Linux:** no Ozzy needed; the kernel's own USB MIDI driver handles it. Could someone confirm MIDI out works? Find the port with `amidi -l`, then e.g. `amidi -p hw:X,0,0 -S '9F 17 7F'` to light a button LED.