mischa85/Ozzy #30: macOS 15.5 issue

BorisUytterhaegen, 2025-07-17:
Hi,

Having trouble getting through the first couple of steps of the install manual for macOS.
Running macOS 15.5 Sequoia, on a Macbook pro with M2 pro silicon.

- To reboot into recovery mode do not press COMMAND+R, that does not work, press and hold the power-button on startup.
- `csrutil disable`  command runs fine.
- When I try running `sudo nvram boot-args="amfi_get_out_of_my_way=0x1"`, it does not recognise the sudo command, sudo is not available in recovery mode so it seems, trying to run the command without sudo, tells me I do not have the right permissions. Sudo is installed on the systems, as I do have the sudo command available in normal operation.

So this is where the adventure stops for me.
Anyone that can help me to the next step?

Many thanks!

Best,
B

BorisUytterhaegen, 2025-07-17:
Perfect, thanks for your reply.
Will try again with the updated instructions!

mischa85, 2026-10-11:
Closing since the instructions were updated. Thanks!

closed 2026-10-11 by mischa85